FabMo/FabMo-Engine #783: implement all [FP] parameters for file-running

RobMackie, 2022-01-04:
Iimplement all [FP] parameters for file-running; some may be done or stubbed in; such as proportion and 2d or 3d offset; 

"the OpenSBP command [VA] sets the current locations (first 6 parameters) and the Table Base Coordinate (G55) offsets. There is a problematic sequence effect when a location and offset are applied in the same command, such as VA,0,,,,,,0. The location is zeroed, then the offset is zeroed, and then -- as effect of zeroing the offset -- the location is changed to accomodate. I -think- that if we do this in the opposite order, the effect will be correct at the end of the process ... both location and offset being zero.
  - helpful to run this in the Sb4 app, checking config after each ... "

tedh-shopbot, 2022-01-25:
To Clarify:  This issue is about implementing advanced file running features in FabMo. The implementation will take the form of: 1) adding functionality to the FP command in the OpenSBP runtime, here duplicating some of the parameter-specified features of Sb3; and 2) adding functionality to the FabMo Job Manager. The functionality will essentially be the same for each, relying on the same function calls at the time a file is being run. At the moment, I expect they will also appear as the same optional file "Transform" sheet.

The gist of this functionality comes from the current functions implemented in Sb3 that extend the FP command that runs a file. These are:
1-Scale-x, scale-y, scale-z
2-repeats
3-3D offset from current location, 2D offset from current location
4-plunge depth for a 2D file
5-add tabs, tabs parameters specified in the VB
6-move to z-zero before plunging

We don't need or want to do all of these.
1. Scaling - YES
2. Repeating - One type, MAYBE, see below
3. 3D and 2D offsets - These can be very convenient, and they are easy to implement, so - LATER, not a priority for first release
4. Plunge - NO, this was useful primarily in the pre-CAM days
5. Tabs - NO, a useful feature for us pre-CAM; there is no practical path to get there with G2 in any case
6. Move to z-zero - NO ... I don't see a need now
*On repeats. We previously used the repeat function 2 ways. 
=One was to support a template type of function where a user would provide a 2D file, and specify a depth, and then cut in several places ... this was pre-CAM
=The second was just to repeat a 3d file a number of times. This is useful in a number of ways:
    -for diagnostics
    -with a stepover at end to "stamp" a file across a sheet
    -with a pause to automate production; sleazy, but some like it
    -just to get a lot of reps to automate something like polishing
So I believe we want this latter repeat capability on the File running wish-list, but for LATER, not required for release

Interestingly, much of what we want here such as getting offering for OpenSBP and for FabMo has been already been done in FabMo under the theme of TRANSFORMS. Some of the planned transforms are already functional, some are done but broken, some are stubbed-in but need fleshing out. Basic scaling of a file in X, Y, and Z is functional, for example.

None of these is easily called up, and no links of the functionality have been made to the Sb4 interface.

So my next step here will be to add details on what we've got, and what is left to do.

tedh-shopbot, 2022-02-15:
Offsetting and Scaling TRANSFORMS are working EXCEPT for arcs.

tedh-shopbot, 2022-02-22:
Check Rob's problem with limits on at start.

tedh-shopbot, 2022-05-06:
Mostly covered with PR #863 including implementing scaling of arcs and circles.

closed 2022-05-06 by tedh-shopbot