FabMo/FabMo-Engine #790: Input driven feedhold during the dwell as spindle spins up during resume causes FabMo to lock up requiring reboot

RobMackie, 2022-01-29:
If you run a job that spins the spindle and during motion you create a feedhold, the spindle retracts and powers off. When you hit resumne in FabMo, the spindle spins up during a "dwell" and then plunges and motion restarts. 

IF you hit the feedhold button while the spindle is "spinning up" and before motion has resumed (i.e. before the STOP button has fully appeared on the user interface) then everything goes wrong with "spinning icons" etc. and the only solution seems to be a reboot of the whole tool. One can't do this via the web interface because the stop button isn't presented until after the dwell is over. Moreover the spindle continues to spin and the only way to stop it is E-stop or power down.

A good fix to this would either ignore the feedhold during dwell (I consider this a bad fix - dangerous) or would make it so the user interface doesn't lock up and a feed hold takes place.

kyle-kingsbury, 2022-02-02:
Possible duplicate of https://github.com/FabMo/FabMo-Engine/issues/786.

kyle-kingsbury, 2022-03-10:
Made a few adjustments to G2 core (https://github.com/FabMo/FabMo-G2-Core/pull/48) which allowed feedholds while resuming, also a few adjustments made in Fabmo. Changes to how input feedholds are handled from PR https://github.com/FabMo/FabMo-Engine/pull/818 were required for this to work properly, specifically the spindle spin up behavior was not being handled correctly. The PR is not merged with master as of me writing this. 

Also added a check in G2 handleStatusReport which checks whether a feedhold is triggered while a resume is pending. This led to a result that was weird, yet functional. The green spinner that replaces the resume button was being displayed, yet if you click on it, the program resumes. I adjusted the FabmoUI to eliminate the green spinner in this scenario. Not a perfect fix, but much better than completely crashing and forcing a tool reboot.

As far as testing I ran all the dev checks without error. I tested dev check one extensively. Stopping, resuming, stopping while resuming, clicking resume multiple times while it is resuming, pressing the input feedhold multiple times before resuming. Everything seems to be working correctly.

kyle-kingsbury, 2022-03-29:
Fixed with PR https://github.com/FabMo/FabMo-Engine/pull/825

closed 2022-03-29 by kyle-kingsbury