FabMo/FabMo-Engine #792: gcode runtime feedhold from software "STOP" can not be resumed sometimes

RobMackie, 2022-02-01:
In the gcode runtime, if "STOP" in the footer is clicked, then motion stops. Some times (pretty frequently) the resume/quit display disappears and the "STOP" footer returns with no user intervention. When this happens, the job does not resume and if the  user clicks "STOP" the infinite spinny arrow happens, and if the user doesn't hit stop, nothing happens. 

The condition only seems to happen on the first "STOP" of a job and doesn't always happen. If it doesn't happen on the first feedhold, it seems not to happen for the duration of that job, regardless of how many feedholds are exectuted.

Work around - use the escape key to send a "kill" '!' to g2core. The job aborts and one can then restart the job.

jkeppeler-parthenon, 2022-02-01:
[image]
Issue seems to be this "jiggerypokery" triggering

kyle-kingsbury, 2022-04-18:
Fixed with https://github.com/FabMo/FabMo-Engine/pull/846. Related to stat 3 causing programs to end early.

closed 2022-04-18 by kyle-kingsbury