FabMo/FabMo-Engine #795: Spindle shuts after feed hold (gcode)

kyle-kingsbury, 2022-02-02:
Input driven, and software driven feed holds during gcode file execution sometimes causes the spindle to remain off. The issue often does not occur during the first feed hold. It seems to happen every time after the first feed hold.

tedh-shopbot, 2022-02-16:
Yes. That is what you mean!
{I'm thinking this should probably be in G2}

Testing with FabMo ver 3.0.89 and G2 101.57.31, 101.57.32, & 101.57.32-4 (all the same)
Using the file:
ms, .5, .5
so, 1, 1
mx, 4
'PAUSE 2
'cc, 2
mx, 0

With the two line commented out: the feed-holds and spindle seem to work correctly for me (4-5 Stops in each leg). Always restarting and showing the appropriate delay.

If I un-comment the circle command, then in the leg leading up to the cc I am able to stop and resume correctly. But once in the circle, or after the circle in the last leg, the spindle consistently fails to restart after a Stop - Resume.

If I un-comment the PAUSE, and I put a STOP just before it, the tool crashes and loses connection to G2.

FINALLY, if I just try STOP-RESUME with the sb_logo file, the spindle does resume even after an arc, BUT there is no spin-up delay! 

This really has me puzzled because I am sure I tested the logo file when I reviewed the earlier pull requests related to this stuff.

tedh-shopbot, 2022-02-16:
And, to round out this story (I was not crazy).
The G2 firmware at the end of the branch for pull request #12 in G2 --> 101.57.27-4 and the firmware after the merge 101.57.28 both work correctly with the above testing.
Thus we lost the feed-hold spindle functionality between 101.57.28 and 101.57.31.

And, I would add, that when doing lots of stops and resumes, it is too easy to get that FabMo crash that is exacerbated by having a PAUSE in the vicinity of the the stop.

RobMackie, 2022-02-16:
This means we need to go to fight the tests more because I'm running all the tests with many stops and resumes. And because I wasn't within hearing of my machine I can't guarantee that the spindle always restarted. But I am not processing pull requests until I've had time to run them through those tests in Dev checks

tedh-shopbot, 2022-02-16:
Let me identify the commit that broke it and I will then post the issue to G2.

tedh-shopbot, 2022-02-16:
Issue moved to FabMo-G2-Core issues and closed here.

closed 2022-02-16 by tedh-shopbot