FabMo/FabMo-Engine #808: Reading a variable input from a PAUSE statement no longer working.

tedh-shopbot, 2022-02-24:
This functionality no longer working. Variable not being read or saved or something ...
[@ FabMo-Engine Master ver 3.0.89 +6]

(Pull Req: 
[image]
)
PAUSE accepts a second optional argument of a variable that will be set to the value returned from the pause window input on resume.
PAUSE "Set travel length", &myvar
PAUSE "Travel set to:  " + &myvar
MX, &myvar

will open the pause modal with the message and an input box.  On resume 
the value from the input box is stored in the &myvar variable then 
displayed in the next pause message as a confirm on resume it will be 
used for travel length of the following motion. Can also be used to set 
persistent "$" variables.

This does not include any validation so is not safed in any way to 
prevent a user from causing errors with bad input.  Setting the above 
variable to 'potato' would not be prevented and lead to an error state 
as would resuming with a blank value (we may have the input set as 
required to resume worth testing and adding if not)

Josh Keppeler

jkeppeler-parthenon, 2022-02-25:
I think this is a race condition.  The next pause will lookup the saved value before the value finishes saving.  This might be an issue we need to just have a warning about (Short timed pause following input pause as recommended OpenSBP pattern) or we need to look into locking the config when updated.  We could have an in memory cache that stores the input value in memory before resuming but that leads to the potential for the memory cache being out of sync with the config values in limited cases on crash following an input.

jkeppeler-parthenon, 2022-03-07:
Verified this occurs when there is a timed pause or lengthy move between entries.  This does not appear to be a race condition.

RobMackie, 2022-03-11:
Verified that the race condition version exists but that if one removes the race condition (don't use back to back pauses that reference the variable) then the function works correctly.

RobMackie, 2022-03-22:
Resolved with earlier PR