FabMo/FabMo-Engine #828: OpenSBP variables are written out asynchronously, wrt/ opensbp parsing, needs to appear synchronous

RobMackie, 2022-03-11:
There is a race conditioni where the opensbp runtime can write out a variable and then move foward such that the variable write has not been completed, thus next steps fail due to inability to access the variable.

Need to improve the sequencing, either by making it synchronous or using some other mechanism to solve this:

Ideas:

best first:
**** we could make it so that opensbp parsing doesn't progress  until the callback reports on the status of  the attempt to write asynchronously.  *** 

Others:
we could create a "fast read-only cache" in memory for variables in the process of being written out. when the callback occurs that says the write is complete the cache is unpopulated. (partial solution - doesn't handle writes and write-thrus would be complicated; and the whole thing seems complicated.

we could make writes fully synchronous locking up node untl they are done.

jkeppeler-parthenon, 2022-04-04:
Looks like this is merged in now and ready to be closed

RobMackie, 2022-04-18:
closed with 828 synch config writes #838

closed 2022-04-18 by RobMackie