FabMo/FabMo-Engine #839: ZT double executes

jkeppeler-parthenon, 2022-03-24:
ZT will double execute and can lead to out of order openSBP execution depending on race conditions.

Relates to #822 

On execution of the ZT command a direct emit_gcode is called directly from the command module.  This is followed by a full config write of zeroed positions.  Upon G2Core completeing the G28.3 command it correctly issues a stat 3.  This stat 3 per our event loop triggers the execute_next function.  This all occurs while the write is taking place.  Upon completion of the write the ZT callback is fired leading to the incrementing of the program counter and the execution of execute_next.  

By the time the write has completed execute next has been called on the same program counter leading to a second call of ZT (and pending incrementing and execute_next)  Leading to a duplication of the ZT execution and further confusion as additional stat 3 and callback driven Execute next behavior can pull us out of pause states or otherwise lead to out of order execution of commands or other program structure.

We will need to either implement some kind of ZT blocking action to the stat 3 based execute next which may get the job done but is not a great pattern.  Or revise the ZT command to file its gcode into the gcode queue for in order execution upon priming.

Lastly I'm not entirely certain of the ZT commands purpose as opposed to Z6 which zeroes the same six axis's without emiting gcode directly.  It may be most straight forward to have ZT be an alias for Z6.

tedh-shopbot, 2022-05-06:
Can't be Z6, which zeros the current machine, but not the g55 offsets. ZT wipes the slate. Should be queued per Josh's next to last suggestion.

RobMackie, 2023-04-28:
fixed with the VA changes. (per Ted)

closed 2023-04-28 by RobMackie