FabMo/FabMo-Engine #848: Implement feedhold and authorization config for pendent

RobMackie, 2022-04-05:
Will link a description of desired behavior here soon. and then assign to someone.

RobMackie, 2022-05-02:
Implement/test feature where the spindle enable key going from off to on is the method of "authorizing" the tool, if "Authorization" is configured.

Note: with the new voltage divider circuit on the spindle contactor coil wired into an input, this appears to work as desired if configured correctly. The extent of the resolution of this bug may simply be to test that that is correct, to decide which input to use, and to document the correct configuration of the software.

kyle-kingsbury, 2022-05-04:
These are the configurations that I used to make some progress.

Auth button input = 4
Interlock Input = 2
quit button = 4 by default, needs to not be 4
input 4 - action = none / polarity = high
input 5 - stop

When Rob and I tried this, we got it working with input 4 set to low. Different configurations can produce the same result I guess. The problem is, every time there is a pause in the program it requires another authorization. Still needs work.

RobMackie, 2022-05-10:
I wonder if #631 might not help with the whole authorization around short timed pauses