FabMo/FabMo-Engine #861: Previewer Pointer obscured by dark grey plane, see pics

kyle-kingsbury, 2022-04-25:
[image]
[image]
[image]

tedh-shopbot, 2022-04-26:
This happens because table position is not set ...
I'm about to post a PR that puts in defaults. Meanwhile go to the setting and put table below zero. (e.g. -0.75)
Also changing the icons to make the info more obvious and consistent.

kyle-kingsbury, 2022-04-26:
Ah alright, that makes sense!

tedh-shopbot, 2022-04-28:
Fixed in PR for "preview_defaults" which sets table below grid by default for metric and inch.

closed 2022-04-28 by tedh-shopbot