FabMo/FabMo-Engine #864: SB4: Not able to seamlessly run part files using FP

kyle-kingsbury, 2022-04-26:
If I try to load an sbp file off my desktop there is an error in the log that the routes are incorrect. You can get a file to run if it is already pending in the job manager, but this does not seem intuitive.

RobMackie, 2022-04-27:
Explaining what was discovered that led to closing this. 

If a job is queued in "Job Manager" and opened in the Editor and "Run Immediately" from the editor, it remains in the job queue until deleted. It does not get recorded as run from the job queue. (everything in history is a job run from the job queue.)  It must be deleted from "pending" in the job queue and then does not appear in the history. If the job is run from the "pending" queue directly (not from editor) the it appears in the history tab lislt.

kyle-kingsbury, 2022-04-27:
I closed this due to my own stupidity. I was loading an aspire file not a .sbp file! The job history issue is detailed in issue https://github.com/FabMo/FabMo-Engine/issues/859.

BrianOwen, 2022-04-27:
Does that mean we don't filter which files can be uploaded to the job manager? That would be helpful in a lot of cases to only show .nc and .sbp files in the file browser.

kyle-kingsbury, 2022-04-27:
That's a great point Brian. Did not even consider that. If it happened to me it will happen to someone else.

BrianOwen, 2022-04-27:
Customers also get confused about the difference between a vcarve file and an sbp when they’re first starting out

Brian