FabMo/FabMo-Engine #875: RPi-Image-22: Network issues

tedh-shopbot, 2022-05-03:
Here's what I have been seeing with the new image. Don't know how much of this is specific to my home network (which is pretty ordinary).

Possible issue with IP reporting in wireless name when using clones of the same SD card, on same machine:
- When I tested a second SD card for the simple first time of ethernet-network on boot. Only the AP-02 name was displayed in the PC wifi report, despite the fact that FabMo was live at the same ip address of the previous SD card and the wifi manager in FabMo showed both the correct ip and the AP mode ip. Then after waiting a day, everything came up showing as expected in the PC wifi report. Could this have something to do with the UUID of the SD card? This did not happen with the first SD card which almost immediately reported the ethernet ip address in the PC wifi name report.

More problematic issue:
- When having started with the ethernet-network and used the FabMo wifi manager to establish a wifi connection. At that point the ethernet connection and wifi connection ip's both displayed in 'Recent Connections' and both connected and ran at their respective ip's. Then a reboot with the ethernet unplugged. No wifi connection existed nor was one reported in the PC wifi report. It looked like it was the password that was not remembered as if I set a password in the RPi Conman, then FabMo came up at the expected ip and would continue to come up after reboots.
- However, the ip was not displayed in the PC wifi report.

*When in either the AP-mode wifi connection or a regular network wifi connection, in the wifi manager the 'Recent Connection' panel does not display. This seems maybe correlated with the absence of display of the ip in the PC wifi report.

tedh-shopbot, 2022-05-03:
I might also note that earlier with this image, I felt like I was getting frequently changing ip when I was direct ethernet connected to the windows PC (169.x.x.x address). The networking in the previous image did not have this problem and it would be a hassle for people wanting to direct connect. I have not been able to duplicate this problem recently but will watch for it as I expect that the direct ethernet connection may be the one of choice in some settings.

tedh-shopbot, 2022-05-06:
Image and NETWORKing:

The issue seems to be with wifi connections
if you are connected via AP or wifi-to-network; THEN there is no display of the Recent Connections panel in Network Manager.
if you are connected via AP or wifi-to-network; THEN there is no display of the IP address in the client wifi name list, though connections exists and seems to work fine. Display of IP in names works pretty reliably for ethernet direct and ethernet to network router.
when a wifi connected, you don't notice anything strange when running in the systemd start, but if you do a dev start you get the following error:
debug: Watching /fabmo/dashboard/apps for changes... [debug]
info: wpa started with: Selected interface 'uap0'
wpa_state=INACTIVE
ip_address=192.168.42.1
p2p_device_address=de:a6:32:2c:93:2e
address=dc:a6:32:2c:93:2e
uuid=84888164-46bd-570a-ad7a-0c56ea73b4bf
[network]
UNCAUGHT EXCEPTION
TypeError: Cannot read properties of undefined (reading '0')
at RaspberryPiNetworkManager.checkWifiHealth (/fabmo/network/linux/raspberry-pi/index.js:162:62)
at Timeout._onTimeout (/fabmo/network/linux/raspberry-pi/index.js:494:14)
at listOnTimeout (node:internal/timers:559:17)
at processTimers (node:internal/timers:502:7)
info: Saving log... [log]
info: Log saved to /opt/fabmo/log/fabmo-1651792819252-log.txt [log]

With the UNCAUGHT EXCEPTION ERROR repeating about every 10s.

obra, 2022-05-15:
FWIW, I'm seeing this issue in the case where I:
* initially connected over wired ethernet with with a link-local address (169...)
* enabled wifi
* turned off the machine that was connected over wired ethernet.

PR #892 (2022-05-17): Bulletproofing against eth0 disappearing
In cases where eth0 was previously connected to the device and it then goes away, we'd be looking at properties of objects that just don't exist anymore.

Here we use javascript optional chaining to short-circuit the failure cases.

At least locally this fixes #875
commit f7e58b9d (2022-05-17): Bulletproofing against eth0 disappearing (#892)

In cases where eth0 was previously connected to the device and it then goes away, we'd be looking at properties of objects that just don't exist anymore.

Here we use javascript optional chaining to short-circuit the failure cases.

At least locally this fixes #875
closed 2022-05-17 by RobMackie