FabMo/FabMo-Engine #876: RPi-Image-22: thoughts for next image

tedh-shopbot, 2022-05-03:
- This differs from previous image in that if a wifi connection is set up FabMo, if FabMo is not running for some reason, the Pi doe not automatically establish a connection. Not sure how essential that is, but I found it sometimes helpful for the networking to seem to exist outside of FabMo as well as inside. 
- Along those lines, I think adding the Conman gadget to the menu bar might be helpful. If you have a screen you can readily check your connections this way. Or, at least previously, one could.
- I think I saw somewhere that there is a method autoexpand the file system on an SD card; it is very easy to forget and end up with a crashie system because there is little memory left with the 8g version of the image.
- We should check the video defaults for the RPi so that the most common HDMI's will work out of the box. (With the older Handibots that use the RPi screen, we need to enable a special video driver and rotate the screen -- maybe something for tech support to know).
- Might be nice for developers to adjust the system limit for 'number-of-file-watchers' for working with vs-code.

RobMackie, 2022-05-03:
This differs from previous image in that if a wifi connection is set up FabMo, if FabMo is not running for some reason, the Pi doe not automatically establish a connection. Not sure how essential that is, but I found it sometimes helpful for the networking to seem to exist outside of FabMo as well as inside.

I think there is already an issue open for addressing this.  I need to add more descriptive text to it but it is: https://github.com/FabMo/FabMo-Engine/issues/837

RobMackie, 2022-05-03:
Along those lines, I think adding the Conman gadget to the menu bar might be helpful. If you have a screen you can readily check your connections this way. Or, at least previously, one could.

yes, i agree.

RobMackie, 2022-05-03:
I think I saw somewhere that there is a method autoexpand the file system on an SD card; it is very easy to forget and end up with a crashie system because there is little memory left with the 8g version of the image.

I think that we need to expand them before we ship them. For engineering purposes, it might be easier to make that a manual process so we don't have to contract them again before making a new image to share.

What has been your process for "shrinking the image back down" after you expand it? (or have you always flashed a new unexpanded image before playing with the image?

RobMackie, 2022-05-03:
We should check the video defaults for the RPi so that the most common HDMI's will work out of the box. (With the older Handibots that use the RPi screen, we need to enable a special video driver and rotate the screen -- maybe something for tech support to know).

Ok - right. we'd so want it to show up on a monitor easily. Have  you had any problem with the current image? just curious. 

I'm not sure we're supporting the handibot out of the gate? or are we? eventually we will want a way to make that easy, for sure. Not sure we need it in a current image that is aimed only at desktop and only at access from the browser. We know there are significant problems if the user runs the browser on the pi. That will likely be an "unsupported" configuration. A user is welcome to try it but not the recommended or supported way to run it. We don't expect to expose the video ports unless you have the electrical enclosure open. It is there for diagnostics, not use? 

Or do you have a different idea of how that goes?

RobMackie, 2022-05-03:
Might be nice for developers to adjust the system limit for 'number-of-file-watchers' for working with vs-code.

Worth thinking about.  There are some tradeoffs. Need to work out if any of them impact us. We might just document how to change that number and how to change it back rather than doing so, in order to conserve system resources under ordinary use.

tedh-shopbot, 2022-05-06:
**On Shrinking:**
All was able to figure out was to make an SD copy on the Pi, but using an 8G sd card. I'm sure there is a better way. But the pi seemed to shrink the size down OK. To that, which I would then ZIP. And, out of laziness, of done 32-ZIPped to 32 several times.

tedh-shopbot, 2022-05-06:
**On Supporting Handibot:**
There are relatively few Handibots than can be easily switched. I don't think it is worth worrying much about at the moment. Since I am the one who created the weird (no-prox) homing for Handibots, I figured that when I got a chance, I would work up the special macros needed for existing Handibots. I think they will need their own Macro 2,3, and 9. I can mostly just port those from before using the new Standard Macro Variables.

tedh-shopbot, 2022-05-06:
Image and NETWORKing:
- The issue seems to be with wifi connections
- if you are connected via AP or wifi-to-network; THEN there is no display of the Recent Connections panel in Network Manager.
- if you are connected via AP or wifi-to-network; THEN there is no display of the IP address in the client wifi name list, though connections exists and seems to work fine. Display of IP in names works pretty reliably for ethernet direct and ethernet to network router.
- when a wifi connected, you don't notice anything strange when running in the systemd start, but if you do a dev start you get the following error:

> debug: Watching /fabmo/dashboard/apps for changes... [debug]
> info: wpa started with: Selected interface 'uap0'
> wpa_state=INACTIVE
> ip_address=192.168.42.1
> p2p_device_address=de:a6:32:2c:93:2e
> address=dc:a6:32:2c:93:2e
> uuid=84888164-46bd-570a-ad7a-0c56ea73b4bf
>  [network]
> UNCAUGHT EXCEPTION
> TypeError: Cannot read properties of undefined (reading '0')
>     at RaspberryPiNetworkManager.checkWifiHealth (/fabmo/network/linux/raspberry-pi/index.js:162:62)
>     at Timeout._onTimeout (/fabmo/network/linux/raspberry-pi/index.js:494:14)
>     at listOnTimeout (node:internal/timers:559:17)
>     at processTimers (node:internal/timers:502:7)
> info: Saving log... [log]
> info: Log saved to /opt/fabmo/log/fabmo-1651792819252-log.txt [log]

With the UNCAUGHT EXCEPTION ERROR repeating about every 10s.

tedh-shopbot, 2022-05-17:
Hey @RobMackie , I think we covered most of everything here. I'd say we just need to save an issue for making a R-pi handibot profile ... assign to me.

obra, 2022-06-14:
It looks like most folks building pi images use https://github.com/RPi-Distro/pi-gen to reliably script the build of an image based on the vanilla Rpi images. 

Thankfully, it's just a bunch of shell scripts with "stages" of build, so the first 80% of the work to build a custom fabmo version *should* just be documenting the steps you'd need to take to customize a regular image in the form of a shell script.

One thing that you'd theoretically get for free by doing it this way is support for Raspberry Pi's "imager" tool that lets users configure usernames, passwords and network from their desktops as they write the image to an SD card.

RobMackie, 2022-06-15:
Thanks for the reference. I'm hoping to start this in a couple fo weeks. tons of stuff between here and there. :-)

obra, 2022-06-15:
I can believe it. It's a tech stack I've been meaning to play with. If I happen to get anything put together that can give you a head start on this, I'll let you know.

RobMackie, 2022-11-04:
revist during or after issue #990