FabMo/FabMo-Engine #895: Linux Image: Auto logon in to UI

tedh-shopbot, 2022-05-16:
Previous images have had the desktop UI set to auto-login. For our use of the Pi (as a server on a tool), I believe this is the most convenient initial behavior for users. My copy of the current image occasionally requires a login, which I take to mean the auto-login is not yet set ... or may no longer be available. If it is available, my preference is to make things as easy as possible and set it on for our users.

Side note: In the early years of FabMo with both the edison and the pi, there were several different ideas about how to set the user and password. Most ended up creating some sort of confusion. By sticking with the single, universally recognized starting name for every raspberry pi, "pi" we found we made our life much simpler. For a password, "shopbot" made sense and we put that on the background screen of the image so that anyone would know that they had an image with that original password.

There are many "better" ways to do this ...

RobMackie, 2022-11-04:
revisit after issue #990

RobMackie, 2023-04-28:
irrelevant to non handibot tools.