FabMo/FabMo-Engine #914: Confusing "OK" dialog

obra, 2022-05-19:
[image]

This should just have an "ok" button or its moral equivalent.

(Also, the text is wrong. I can't hit keys. I have to click one of the buttons.)

tedh-shopbot, 2022-06-03:
More options for that dialogue have been on the wish list for a while, and a simple "ok" is one. I can't remember how far JoshK got in his work on them. @RobMackie may know. Once it is available, that dialog call can be modified. In any case, the dialogue is there primarily as a test to make sure that a terminal dialogue will display should a user want to program one in that position. Over the years, I've used that that logo file as a first pass test case ... so it has a lot of useless debris floating around in it.

RobMackie, 2023-04-28:
Repurpose this bug. We need pause to have parameters that allow us to specify which buttons appear. Need to change OpenSBP to support this.