FabMo/FabMo-Engine #923: PR #901caused a regression. The collection process for the shopbotsupport.zip file no longer populates the zip file.

RobMackie, 2022-06-06:
#901 - For whatever reason, putting a thumbdrive in with a shopbotsupport.txt trigger file now generates an empty shopbotsupport.zip file. Wondering if it was verified by looking for a zip file and not by opening the file to look for contents. Need test cases with documented pass/fail requirements for all of this stuff.

kyle-kingsbury, 2022-06-06:
Just tried it, fabmo.html, network.txt and shopbotsupport.txt are all present and populated. Shopbotsupport.zip is created but there is nothing inside.

obra, 2022-06-08:
Yikes. I'm so sorry. I'm presuming that It's *something* around these two lines.

```
    chdir($TEMP);
    system( "/usr/bin/zip", "--recurse-paths", "--quiet", "shopbotsupport.zip",
        $REPORT_ID );

```

When I was testing, I did actually verify that the content got into the zipfile, but I *was* running the process by hand, rather than letting systemd notice the usb plug events.

obra, 2022-06-08:
Testing this out in a docker image (because I can't access my Pi right now):

The files it can't access make sense, since the docker image doesn't have those files, but it *does* generate the zipfile with content inside it.

```
root@b0a63e1877ae:/fabmo# mkdir /tmp/disk
root@b0a63e1877ae:/fabmo# touch /tmp/disk/shopbotsupport.txt
root@b0a63e1877ae:/fabmo# ./bin/collect_and_export_data_to_usb_drive /tmp/disk/
/bin/cp: cannot stat '/var/log/daemon.log': No such file or directory
/bin/cp: cannot stat '/var/log/syslog': No such file or directory
dmesg: read kernel buffer failed: Operation not permitted
umount: /tmp/disk/: must be superuser to unmount.
root@b0a63e1877ae:/fabmo# cd /tmp/disk/
root@b0a63e1877ae:/tmp/disk# ls
shopbotsupport.txt  shopbotsupport.zip
root@b0a63e1877ae:/tmp/disk# ls -lart
total 20
drwxr-xr-x 2 root root 4096 Jun  8 03:29 .
-rw-r--r-- 1 root root 8132 Jun  8 03:29 shopbotsupport.zip
-rw-r--r-- 1 root root   71 Jun  8 03:29 shopbotsupport.txt
drwxrwxrwx 1 root root 4096 Jun  8 03:29 ..
root@b0a63e1877ae:/tmp/disk# unzip shopbotsupport.zip
Archive:  shopbotsupport.zip
   creating: fabmo-cb4bdc1481dd420bae78a0ef98948fdf-2022.6.8032902/
  inflating: fabmo-cb4bdc1481dd420bae78a0ef98948fdf-2022.6.8032902/fabmo-1653673860078-log.txt
 extracting: fabmo-cb4bdc1481dd420bae78a0ef98948fdf-2022.6.8032902/dmesg
  inflating: fabmo-cb4bdc1481dd420bae78a0ef98948fdf-2022.6.8032902/ifconfig_out.txt
  inflating: fabmo-cb4bdc1481dd420bae78a0ef98948fdf-2022.6.8032902/system_status.txt
root@b0a63e1877ae:/tmp/disk#
```

I'm going to have to test this on my pi again to see where it's falling down.

RobMackie, 2022-06-08:
ok. i'll investigate. I think I'll get to it tomorrow. not a tragedy. just something to fix.

obra, 2022-06-14:
I've just updated my Pi to fabmo v3.1.26. I then inserted a USB stick containing only the file shopbotsupport.txt. 

Everything sure *looks* like it worked right. This is the file that the pi created.
[shopbotsupport.zip](https://github.com/FabMo/FabMo-Engine/files/8904513/shopbotsupport.zip)

@RobMackie, @kyle-kingsbury if it's failing for both of you, I suspect there's a difference in our rpi images that the script is bumping up against. 

Can one of you try to test the tool manually for me? This recipe should let you do so:

```
$ mkdir /tmp/x
$ touch /tmp/x/shhopbotsupport.txt
$ /fabmo/bin/collect_and_export_data_to_usb_drive /tmp/x
$ ls -al /tmp/x
```

This is what that looks like for me:
```
pi@fabmo:/fabmo/bin $ mkdir /tmp/x
pi@fabmo:/fabmo/bin $ touch /tmp/x/shopbotsupport.txt
pi@fabmo:/fabmo/bin $ ./collect_and_export_data_to_usb_drive  /tmp/x
umount: /tmp/x: not mounted.
pi@fabmo:/fabmo/bin $ ls -al /tmp/x
total 124
drwxr-xr-x  2 pi   pi     4096 Jun 14 19:46 .
drwxrwxrwt 16 root root   4096 Jun 14 19:46 ..
-rw-r--r--  1 pi   pi       71 Jun 14 19:46 shopbotsupport.txt
-rw-r--r--  1 pi   pi   112873 Jun 14 19:46 shopbotsupport.zip
pi@fabmo:/fabmo/bin $
```

My best case scenario is that the script explodes on you with an error that will let me solve the problem.

My second best case scenario is that you get a good zip file that you can share with me so I can compare your installed system with mine.

kyle-kingsbury, 2022-06-14:
@obra The .zip file should contain several files for tech support purposes. Yours appears to be empty. There should be 4 files I believe.

obra, 2022-06-15:
That zip file contains a fabmo-[uuid] directory with these files in it. 
[image]

I’ve just tested unzip on both linux and ios. What tool are you using to unzip?

kyle-kingsbury, 2022-06-15:
Hm that's interesting. You are right, if I unzip it the files are there. Before the refactor the files would show up even if you did not unzip the file. Now I see everything that I would expect to see!

obra, 2022-06-15:
> 
> 
> Hm that's interesting. You are right, if I unzip it the files are there. Before the refactor the files would show up even if you did not unzip the file. Now I see everything that I would expect to see!
Oh, right. IIRC, the old script was using the destination as a temp directory, which isn’t a great thing. So (if I’m remembering right) the files ended up next to the zip file because they weren’t getting cleaned up correctly. We’re now using a proper temp dir that’s cleaned up on exit.

RobMackie, 2022-06-15:
I'll try again. I ended up with an empty zip file on the USB  last time I tried. (And several other times. I like that it now uses a temp for properly, just need to figure out why I have and empty final result. I'll let you know

kyle-kingsbury, 2022-06-15:
I tried it on my pi and it worked. The .zip file appears empty until you unzip it.

RobMackie, 2022-06-15:
Ok. so may be we have a zip file compatibility issue. Why would windows see
a standard zip file as empty until unzipped. Most of our customers and
maybe even our tech support people will expect to be able to double click
on the file and see the contents. Is it a version of zip? is an option that
has changed?  it used to work that way. Weird. I'll play with it. It hasn't
gotten to the top of my stack yet, but i guess it's time to see what's up.

Rob.

obra, 2022-06-15:
On Tue, Jun 14, 2022, at 6:42 PM, Rob Mackie wrote:
> Ok. so may be we have a zip file compatibility issue. Why would windows see
> a standard zip file as empty until unzipped. Most of our customers and
> maybe even our tech support people will expect to [...]

Message ID: ***@***.***>That's super weird.  

And yes, I can now repro this in a Win 11 VM. And I see the issue. Patch incoming.

obra, 2022-06-15:
Testing locally, I'm no longer able to reproduce the failure with the fix in #943

RobMackie, 2022-06-15:
Wow. cool! thanks! I'll be testing and merging PRs tomorrow. I'll get this in.

PR #943 (2022-06-17): Clean up data export directory name
The way we were constructing the debugging dump's directory name didn't
properly drop trailing newlines, resulting in a zip file windows didn't
accept.

Add zip to our Dockerfile to enable testing of collect_and_export_data_to_usb_drive

Fixed #923
commit 7f710bbc (2022-06-17): Clean up data export directory name (#943)

The way we were constructing the debugging dump's directory name didn't
properly drop trailing newlines, resulting in a zip file windows didn't
accept.

Add zip to our Dockerfile to enable testing of collect_and_export_data_to_usb_drive

Fixed #923
closed 2022-06-17 by RobMackie