FabMo/FabMo-Engine #925: Document why fabmo can't reconnect to g2 after boot

obra, 2022-06-08:
Right now, if the g2 board and the fabmo controller aren't started up together, fabmo isn't able to connect to a running g2 board.   In practice, I'm finding this mildly frustrating. Is there a fundamental technical reason the two need to be started together or is there just a bit of reconnect code that hasn't been written yet? 

Is there some documentation I should be reading?

RobMackie, 2022-06-09:
I haven't investigated much yet. When it's installed inside a tool sharing
the power switch, it's not much of an issue. But I agree, as a developer
it's a pain.

I do not believe there is any technical reason we can't do it. A starting
place for you to look might be the firmware upgrade. We reboot the g2 card
during an upgrade. (using bossa -R? or maybe a json command. I haven't
looked, yet, even though I have wanted to on several occassions - just
other priorities still) And it successfully reconnects every time. Try it.
You can find old or new g2 code in the fabmo-engine history in git.  You'll
want to work with the most current but installing an older version using
the config page firmware upgrade facility is a worthwhile thing to try.
You'll see what happens and can trace it if you like.

Rob.

RobMackie, 2023-04-27:
Duplicates #948