FabMo/FabMo-Engine #926: Trying to z-zero in DT Fireup just..doesn't do anything.

obra, 2022-06-08:
Clicking the button doesn't generate any response in the UI.

[image]

In the browser's log, I see:
[image]

Other moves with the control pad work fine.

After a manual move, I can z home:

[image]

obra, 2022-06-14:
I'm unable to repro as of right now. I have not yet updated fabmo, but am about to.

Software Version
47884-v3.1.8-dev
G2 Firmware Version
v101.57.36

Please feel free to close this or tag it in whatever way you folks use for "customer can't prove the problem happens" (or ask me to do the same. I just don't know what your standard practice is)

kyle-kingsbury, 2022-06-14:
Rob,
How do we want to handle this?

RobMackie, 2022-06-15:
I figure if we're still trying to reproduce it we'll tag it "reproduce" and
that means if anyone ever reproduces it, we'll give it attn, otherwise it
just sits as a "apparently seen once, may have been fixed, may have had a
transient cause or  may be user observation error - won't know until we see
it again."  If it last long enough (no idea how long yet) without being
reproduced, we'll close it as "reproduce" or if we can figure out it was a
misoberseveratioin (i.e. if jesse says - oh is see what i did, it didn't
really fail, i just misunderstood ) - then we'll mark it junk and close it.

Does that sound about right?

And Jesse - we're figuring this out live as we go. there were one set of
practices and now new ones are emerging. So opinions and suggestions
welcome, sometimes adopted, sometimes not.

Rob.

obra, 2022-10-11:
> 
> 
> That is strange. When I run it output 4 turns on, the stop button appears in the footer, I can see the z raise up and slowly lower until it is below 0. Then I get an alarm "Target Not Triggered! (Started above prox switch?)" (I am running on a simulater with no prox switch). When you ran it nothing happened at all? What is your FabMo software version and firmware version?

I'll pull versions off the mill when I'm next down in the shop, but both are the as-shipped versions you sent out.  It doesn't fail reliably. It just seems to get stuck sometimes.

closed 2022-10-25 by RobMackie