FabMo/FabMo-Engine #929: Job manager's relative timestamp for jobs is wrong

obra, 2022-06-09:
This is for a job I just generated a few minutes ago.

[image]

kyle-kingsbury, 2022-06-15:
That is very odd. Never seen that one before! Keep an eye on it and post here if it happens again, ill do the same.

obra, 2022-06-15:
> 
> 
> That is very odd. Never seen that one before! Keep an eye on it and post here if it happens again, ill do the same.

One possible guess (not based on looking at the code): If that timestamp is based on a pi-side date but converted to a relative timestamp by client-side js and the pi can't get to the public internet, its clock could be wrong, resulting in that kind of skew.

obra, 2022-07-18:
This problem *mostly* goes away when the pi is connected to the internet. But only mostly. In this case, my laptop's clock is a couple seconds off from the pi's, letting me execute jobs from the future.

[image]