FabMo/FabMo-Engine #941: Smooth Sketch: no obvious way to set feeds and speeds

obra, 2022-06-14:
Does Smooth Sketch have a way to change the feeds and speeds? If not, it might be worth having the UI say what the assumptions are so that users don't use settings intended for pine on an ironwood board.

kyle-kingsbury, 2022-06-14:
I tried adjusting the "movexy_speed" setting found in the FabMo configuration app and that had no effect. It seems like the app just defaults the feed rate to 60 IPM. I agree it should give you the option to configure it within the app.

obra, 2022-06-15:
On Tue, Jun 14, 2022, at 6:48 PM, Rob Mackie wrote:
> 
> 
> Most of the apps are still someone's "proof of concept". We're going to
> need to "grow them up" or present them as such. One of the open topics of
> discussion right now is which apps should ship with the first release. My
> take is a very minimal set that can grow as we make them robust enough to
> make customer happy rather than frustrated.Message ID: ***@***.***>
.
As a customer, I'm 100% on board with that plan.

(Also, "here's a new app" every month or quarter is a convenient thing to be able to market or to use to encourage folks to upgrade.)

RobMackie, 2022-06-15:
>As a customer, I'm 100% on board with that plan.
Glad that resonates

>(Also, "here's a new app" every month or quarter is a convenient thing to be able to market or to use to encourage folks to upgrade.)
I'm so onboard that train.