FabMo/FabMo-Engine #954: Dev checks require a browser refresh after running dev-build.sh

kyle-kingsbury, 2022-06-20:
If you run the dev-build.sh script, then try to go to dev-checks you have to refresh the browser before FabMo will allow you in. A minor, but strange, inconvenience.

kyle-kingsbury, 2022-06-21:
In context.js, this is being triggered

		try {
			if(hash && hash != this.engineVersion.hash) {
				console.info("Hard refresh of app " + id + " because hash " + hash + " doesn't match " + this.engineVersion.hash);
				hard_refresh = true;
			}

kyle-kingsbury, 2022-06-21:
This is being triggered when you click on an app after running the dev-build script:

try {
    if(this.engineVersion.debug) {
        hard_refresh = true;
        console.info("Hard refresh of app " + id + " because debug mode.");
    }

RobMackie, 2022-10-25:
not needed

closed 2022-10-25 by RobMackie