FabMo/FabMo-Engine #973: If Z lift is less than current z value, lift does not occur.

kyle-kingsbury, 2022-07-14:
The first cut depth in this file is 0.740 inches. When z lift is set to 0.500", and a feed hold is issued, z lift does not happen. When z lift is set to 1.000", and a feed hold is issued, z lift occurs as it should. 
[Pocket 1.txt](https://github.com/FabMo/FabMo-Engine/files/9114068/Pocket.1.txt)

kyle-kingsbury, 2022-07-14:
I think the intended behavior is to get the cutter off of the workpiece. If I am cutting at z1.000" and my z lift is set at 0.500, then I would expect it to raise to z1.500".  Treat it as current position + z lift instead of treating z lift as the coordinate to move to.

RobMackie, 2022-07-15:
> I think the intended behavior is to get the cutter off of the workpiece. If I am cutting at z1.000" and my z lift is set at 0.500, then I would expect it to raise to z1.500". Having it not lift in certain situations seems odd. The relative lift makes sense to me. The "risk" is that it wont lift high enough. When I wrote this bug I was confused what was happening because the tool did not lift at all. If it did not lift high enough, I would have figured out that my z lift was not set high enough a bit quicker.

I think I like that definition of correct behavior as long as we check for the working volume and never lose steps against z-max.

RobMackie, 2022-10-25:
this is working as designed.

closed 2022-10-25 by RobMackie