FabMo/FabMo-Engine #977: Exposing homing/zeroing  while in Job Manager

obra, 2022-07-19:
I find myself mostly living inside the job manager these days, but find myself constantly having to jump out to another bit of UI to zero a router bit or to zero x & y.  My suspicion is that these operations are common enough that they merit buttons inside the job manager.

BrianOwen, 2022-07-19:
Jesse,

I think this might tend to get overlooked because during testing we often (or I often) zero axes from the manual keypad which can be done as a pop up from anywhere.
What would people think about having those buttons on the keypad? We have that in SB3 with shortcuts for C3, C2 and JH on the keypad window

Brian

obra, 2022-07-19:
That would indeed be a fine place for those features.

RobMackie, 2023-04-28:
let's do this with applets in the DRO