FabMo/FabMo-Engine #1041: Spindle plunges into deck hard on job start

RobMackie, 2023-02-07:
Ran a job on a desktop in the desktop profile. There had been previous jobs. The job was in metric. When the job started, it did not seem to follow the file's instructions.  It plunged the bit hard into the deck as fast and hard as it could.

One suspicion is that the first move was to go down to the safe height before moving to (0,0) in x,y. The vertical move may have been executed in the wrong units. If it tried to go 25.4 times as fast and far as it was supposed to, you'd get an exact match for the behavior we saw.

Will try to get the previous job and the job from history and put them here.  Not sure if I will be able to identify those files, but will try.

kyle-kingsbury, 2023-04-26:
Duplicat of #1044

kyle-kingsbury, 2023-07-27:
At first this just seemed like a duplicate of https://github.com/FabMo/FabMo-Engine/issues/1044. However, since inches vs millimeters was suggested as the culprit, I think it needs to be explored further.