FabMo/FabMo-Engine #1042: Lockup that happens after hitting a physical stop

tedh-shopbot, 2023-02-07:
This happens rarely and I have not been able to get it to reproduce reliably.
Early in a session (typically) if a tool (DT or Handibot) runs into a hard stop, FabMo locks up ... as if it may have lost communication with G2. I have not been able to confirm what actually happens. Perhaps it is only in manual mode?

It is not related to inputs being triggered, because Handibot has none. My intuition is that it is caused by an electrical supply effect related to changed load on a stepper motor and driver, spike or brownout.

kyle-kingsbury, 2023-04-17:
Saw this today for the first time. Hopefully we can mitigate it with limit handling someday.

tedh-shopbot, 2023-04-17:
It does not happen often, but seems to be there. I have seen it on the Handibot I use for testing (has the same electronics as the DT). I have been running for several weeks with a simulator plugged into the Handibot with a 10ft USB cable. I have not seen that behavior at all. Perhaps it is because the power to the Pi is better isolated? Or, just proximity??