FabMo/FabMo-Engine #1067: Enhancement: Make more features keyboard capable

kyle-kingsbury, 2023-04-05:
A specific request is using < and > to adjust the move speed while in the manual keypad. Another example would be toggling fixed mode in the manual keypad with a hotkey.

tedh-shopbot, 2023-04-11:
I like those. I would add clicking anywhere outside to close the Keypad.

tedh-shopbot, 2023-04-17:
Yes. Even though all the ones mentioned to date have been implemented, it would be useful to additionally include the keys for GoTo and Set for the manual Keypad ... and perhaps the axis letters as selectors. Maybe also zero when an axis is selected.

We should do similar work on messages to the user such as Quit/Resume.

kyle-kingsbury, 2023-04-17:
The two original enhancements were added in https://github.com/FabMo/FabMo-Engine/pull/1072. 
Some others for consideration are: 
- XYZ keys could select one of the location entry windows for data entry. Once in that selected state:
- ctl-Z could zero the axis (ctl to make sure you really want to zero)
- G could start the Go To behavior
- S could do the Set To behavior
-We could also consider a typematic feature in the fixed mode where holding down long enough would start a repetitive fixed move pattern of punctuated small moves
And, we could consider enabling the Mouse-as-Remote-Pendant per the way Sb3 works