FabMo/FabMo-Engine #1136: Z-lift on "feedhold" or "pause" - lift to z-limit switch as an option, not just to a set height from current zero.

RobMackie, 2023-10-23:
Ted - is still a thing that makes sense to you for the software? Either as the "way it works" or as a configurable option?

If so, then over the next 6 or 8 weeks I can probably find some time to implement it. Might be sooner if things go well in the rest of my life in terms of schedules. 

I'd love to have it work that way, so I'm motivated to do it. My schedule is a bit jammed up, but I'm likely to do it in my own fork anyway, so I figured I'd put this here and if you say you'd like it, then I'll make sure it gets to you. If you'd like to see it done, then assign it to me.

RobMackie, 2023-11-08:
I was looking at this recently and realized that the code for probe aborts on feedhold (or that's how i remember it).  I wonder what will happen when I program in a probe in the middle of a feedhold? I guess I'll find out. If you have thoughts, would love to hear them.

tedh-shopbot, 2024-01-20:
Sorry, just noticed this Rob. In theory, G2 allow some actions within a feedhold by creating a secondary planner. I can't remember offhand if the probe command is one that is supported. Should be interesting to take a look at it. It would be awkward in the OpenSBP runtime handling of a feedhold on the client side.